jrsteensen/OpenHornet #332: Evaluate options for cockpit entry

jrsteensen, 2021-08-21:
### Summary:
Evaluate methods of adding motion to the consoles to allow ease of entry/exit.
Some possibilities may include:

1. Seat Sliding Aft out of the pit and sliding back forward
2. Subfloor allowing consoles to pivot outboard and lock into place with a single pin
3. Subfloor allowing consoles to slide on rails aft and lock into the flight/stowed position with a single pin.

## More Information
### Category:
- [X] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
### Type:
- [X] Feature Enhancement
### Applicable End Item:
- [X] Top Level Assembly

L-Walker, 2021-10-18:
Have we solved this issue with the electromagnetic locking system?

jrsteensen, 2021-10-18:
Yup.

closed 2021-10-18 by jrsteensen