jrsteensen/OpenHornet #340: Compare and decide on resin 3D printer baseline model for OH.

jrsteensen, 2021-08-30:
Compare the following resin based 3D printers for suitability considering affordability, availability, materials cost and availability, resolution and build volume for printing buttons, handles, and other parts with a high surface finish requirement.

- Phrozen Sonic Mini
- Anycubic Photo Mono
- Elegoo Mars 2 Pro
- Elegoo Saturn
- Phorzen Sonic Might X
- Anycubic Photon Mono X

Feel free to chime in on this issue with any other contenders!

ajmilner85, 2021-08-31:
Made a quick comparison spreadsheet:

https://docs.google.com/spreadsheets/d/1fyUyvYBSvN_-gbWMGVPNUXbLbn__Ay_zhoWjUo8PrdA/edit?usp=sharing

Hope it helps!

jrsteensen, 2021-09-02:
I think the [Anycubic Photon Mono](https://www.anycubic.com/collections/anycubic-photon-3d-printers/products/photon-mono-resin-3d-printer) is going to be our minimally viable resin printer for this project.

closed 2021-09-02 by jrsteensen