jrsteensen/OpenHornet #346: UFC Display light channels and dot filter layer

Alan-JH, 2021-09-05:
### Summary:
The current PCB design for the UFC top display layers makes the light too dim, when filled with clear resin, and only allows a few degrees of viewing angle without resin. With resin, the display can be viewed about 15-20 degrees off axis, with some readability. We have discussed making the PCB's drill holes larger, making the PCB thinner, increasing space between the top layer and the LEDs by making the channel layers taller, and making the top PCB black. Laser etching has also been brought up as an alternative.

The current design has the holes spaced 0.9mm apart with 0.6mm diameters. JLCPCB's drill size tolerance is +0.13mm/-0.08mm, so with the holes in their current center to center positions, we could probably only get away with 0.1mm or so extra diameter, which wouldn't make much of a noticeable difference. For thickness, JLCPCB offers down to 0.6mm without a significant increase in price.

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [x] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
ufc_v4_scratchpad_layer4.kicad_pcb
ufc_v4_opt_layer4.kicad_pcb
ufc_v4_comm_layer4.kicad_pcb

jrsteensen, 2021-09-08:
This item will be placed on hold until tested with a "pure" OH-build. Based on feedback from @Mangy_Dog we are (mechanically speaking) doing the same as his successful digits. We need to evaluate it with the OH spec LEDs, and OH spec driving electronics.

jrsteensen, 2022-09-23:
Went different way with UFC. This change will not be implemented at current time.

closed 2022-09-23 by jrsteensen