jrsteensen/OpenHornet #347: Add Fourth Mounting Point for Skew AMPCD

Lazarus94, 2021-09-08:
### Summary:

The AMPCD Center Pedastal Mount is only secured to the wooden frame by three mounting points. This results in it being slightly askew.

### Expected Results/How It Should Work:
A fourth mounting point should be added to properly square the AMPCD. 

### Actual Results/How It Does Work:
With the fourth mounting point added there will be no room for the AMPCD to be skew. There is an issue with clearance from the screen for the Standby Instrument panel, this will be solved once the panel is made fully analog. In the meantime, the fourth mounting point will have to go into the roof of the LIP. See second image.

### Screenshots:

[image]
[image]

### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-09-21:
@Lazarus94 Can you define as to what "slightly askew" specifically means so I can further evaluate this issue?

jrsteensen, 2021-09-21:
@Lazarus94 Not to come off as a broken record, but can you be more specific? Some pictures, measurements, etc? Three mounting points should restrain all degrees of freedom.

Lazarus94, 2021-09-21:
@jrsteensen Not at all. Sorry was on mobile.

Whilst having 3 mounting points does fully secure it there are two on the left side (front facing) and two on the right. The right sides mounting point is at the bottom corner which leaves the top right corner loose. The print is flexible to a degree and the pressure from the other mounting points seems to push it out a bit. See the images below. Its hard to tell in the photos but the Pedestal is at an angle. 

[image]
[image]
[image]
[image]

russellmm, 2021-09-21:
Are you saying to move Screw at location #1? One thing I did was add the panels to Right/Left inboard footwell walls prior to installing those inboard footwell walls to the LIP structure. This ensured that the alignment of those 2 pieces was correct once screwed into the "upper instrument panel mount"

The only area I am see a little flex is at location #2 as that panel is allowed to float there. But that may simply be because I need to print that piece with more infill to make it stronger.

[image]

Lazarus94, 2021-09-21:
@jrsteensen @russellmm I've just had another look and it was my error! I made a mistake when installing the Pedestal. Sorry about that I think the issue can be closed.

jrsteensen, 2021-09-21:
Sounds good! Thanks for the update Laz!

closed 2021-09-21 by jrsteensen