jrsteensen/OpenHornet #348: FDM vs Resin parts. Heat inserts

Ben-F111, 2021-09-09:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
The JETT button and bleed air knob are designed for heat inserts. 

### Expected Results/How It Should Work:
This works great fir FDM. But if resin printed the cavity for the brass insert needs to be just slightly larger than the insert and epoxy or resin glued in. The holes can be currently enlarged by drill. But the resin chips easy. ESP the hole on the outer edge of the JETT

### Actual Results/How It Does Work:
Can 2 types be produced in the freeze. 

And FDM and a resin. For these 

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-09-09:
Can they be pressed in before cure?

Ben-F111, 2021-09-09:
[image]

Ben-F111, 2021-09-09:
[image]

Ben-F111, 2021-09-09:
> Can they be pressed in before cure?

No they cant, there is no give, even before UV

russellmm, 2021-09-09:
I don't have a resin printer (yet) but would this type of insert possibly work https://www.mcmaster.com/screw-to-expand-inserts/. Or make the hole larger and expoy it into place?

Ben-F111, 2021-09-09:
you cant put pressure on resin, there is no give, its like a hard plastic or "glass" it shatters / chips, it needs to be glued from what I have found

jrsteensen, 2021-09-09:
This project has made me discover a hatred of simple knobs. 😅

troyfrezze, 2021-09-13:
I have found I can compress the heat set inserts prior to curing using a clamp to press. It's not caused any damage to my prints and once cured, is solidly in place.  I've not tried with the knobs yet, but plan to print them this week and test.

Ben-F111, 2021-09-13:
I would like to see. But I would be worries about the detail in the opposite side. Ie the JETT text. If you can force an insert. You can surely deform the text. Which is the entire reason you resin printed in the first place. Please let us know how you go.

jrsteensen, 2021-10-31:
Fixed bleed air AUG knob in referenced commit.

jrsteensen, 2021-10-31:
On jett button, added a slot to the straight edge cut, since the wall side of the hole penetrated the edge.

[image]

jrsteensen, 2021-10-31:
Closed with referenced commits.

closed 2021-10-31 by jrsteensen