jrsteensen/OpenHornet #353: Heads Up Display (HUD)

jrsteensen, 2021-09-23:
### Summary:
Design a variable focal length heads up display for OH baseline.

- Should be designed without modification to current UFC.
- Should be designed for variable focal length to keep focus on primary out-the-window display instead of collimated. 
- Integrate AoA indicator to HUD Frame

## More Information
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [X] Software - Sketch
### Type:
- [X] Feature Enhancement
### Applicable End Item:
- [X] Main Instrument Panel (MIP)

jrsteensen, 2021-09-23:
Conceptual Light Path:
LCD -> First Plane Mirror -> PCX Lens (200mm DIA, 600mm FL) -> 30/70 teleprompter glass.

If anyone has any optics experience, I'd love to chat!

kbastronomics, 2023-02-13:
was looking at this and found the lens on ED for reference.

https://www.edmundoptics.com/p/200mm-dia-x-600mm-focal-length-pcx-condenser-lens/1020/

jrsteensen, 2023-05-31:
Hud idea: https://www.headsupview.com/sc1-specs

mietek, 2025-07-19:
> If anyone has any optics experience, I'd love to chat!

Check out this [DIY HUD by João Cabeleira](https://voltaico.net/projects.html).  Here’s a [reddit discussion of it](https://www.reddit.com/r/hoggit/comments/pcxpt2/diy_headup_display/).

PaulR12345, 2026-10-05:
I could possibly help, just give me a while. Also what is more important for you, eyebox range or image accuracy?