jrsteensen/OpenHornet #360: UIP Inboard longeron DXF size/scale

madmoney99, 2021-09-28:
It looks as though the exported DXF files are 1/2 scale. The top of the longeron shows 13.5" length in SW21 but only 6.25" in the dxf file.  The rest of the files appear to be correct but both L and R inboard longerons are sized incorrectly. 

This is in the beta 1.0 branch under release/UIP/Manufacturing/CNC/.75

TonyG on discord.

jrsteensen, 2021-09-28:
Are you opening the DXF as mm or inches? And those files are still the pre-design - we have not created **any** drawings or manufacturing pursuit to beta1.

madmoney99, 2021-09-28:
Inches or mm I see the same values. I understand it’s still in beta/predesign, but there are dxf files that are currently available under the beta 1.0 release branch in git that have incorrect values relating to the UIP inboard longerons.

jrsteensen, 2021-09-28:
Understood, those are holdovers from the master branch and have not been modified. They will be updated post-design freeze. Anything in the release folder of the beta1 branch is obsolete. You should not build to them and expect to be building towards the design freeze.

jrsteensen, 2021-09-28:
Closing this as already part of #314.

closed 2021-09-28 by jrsteensen