jrsteensen/OpenHornet #362: Seat Decals

jrsteensen, 2021-09-30:
### Summary:
Issue to govern seat decals.

- [ ] TBD

jrsteensen, 2021-10-04:
### To Be Designed
@insurgus

- [ ] OH3A2A-9009 - DECAL, AIRCRAFT MOD
[image]

- [ ] OH3A2A-9010 - DECAL, SEAT, AIRCRAFT EJECTION
[image]

- [ ] OH3A2A-9011 - DECAL, SEAT APPLICATION
[image]

- [ ] OH3A2A-9012 - DECAL, MANUAL OVERRIDE HANDLE
[image]

- [ ] OH3A2A-9013 - DECAL, SEAT ARMING LEVER
[image]
[image]

- [ ] OH3A2A-9014 - DECAL, SURVIVAL KIT, OXYGEN, AIRCRAFT SEAT
[image]

- [ ] OH3A2A-9015 - DECAL, PARACHUTE ROCKET
[image]

- [ ] OH3A2A-9016 - DECAL, DROGUE ASSY
[image]

insurgus, 2021-10-05:
The following are included in pull 370.

OH3A2A-9010 - DECAL, SEAT, AIRCRAFT EJECTION
OH3A2A-9011 - DECAL, SEAT APPLICATION
OH3A2A-9014 - DECAL, SURVIVAL KIT, OXYGEN, AIRCRAFT SEAT
OH3A2A-9016 - DECAL, DROGUE ASSY

jrsteensen, 2021-10-06:
Parachute dimensions:
[image]
[image]

jrsteensen, 2021-10-06:
Aircraft Mod Decal Dimensions (rough)
[image]

jrsteensen, 2022-09-10:
Completed with enough for B1HWF. Will add more over time, but not a blocking issue for B1HWF.