jrsteensen/OpenHornet #364: Project Image Renders

insurgus, 2021-09-30:
Requirement for website work

Request the following

- all renders currently used to be exported in 4k resolution (either width or height)
- No "boxed" images. Anything that is cut off in the frame is unusable.
- No shadow in the images
- Transparent background
- No watermarks. They will be embedded at a later time.
- Please no glossy surfaces. Anything showing "light box reflections" should be changed to a matte appearance.

Additionally requesting a specific render of the cockpit from the perspective of the pilot looking out the HUD. This should show all displays. Image 2 on the main website slider is an example of what we're looking for. This should be panned up such that the view would more resemble the pilot flying. Please keep the green on the displays so that they can be masked.

jrsteensen, 2021-09-30:
On hold until beta1 HW freeze.

jrsteensen, 2021-10-01:
Should we also choose a handful of end assemblies and the top level assemblies to do renders at in a manner to do a 360* rotation? (I.e. drag a slider, and the "assembly" rotates 360 degrees?)

jrsteensen, 2023-06-03:
Renders TODO:

- [ ] TL
- [ ] UIP
- [ ] LIP
- [ ] CT
- [ ] LC
- [ ] RC
- [ ] Seat
- [ ] Stick
- [ ] Throttle

jrsteensen, 2023-06-11:
Completed.

closed 2023-06-11 by jrsteensen