jrsteensen/OpenHornet #369: Add Buttkicker power amplifier to console

L-Walker, 2021-10-05:
### Summary:

- [ ] Obtain CAD model for Kenwood X502-1 power amplifier.
- [ ] Add Kenwood X502-1 power amplifier to one of the consoles.

Link to product page: https://www.kenwood.com/usa/car/excelon/x502-1/

Amplifier chosen as discussed due to its worldwide availability, 12V power in, ability to connect remote volume knob, and class D efficency.

## More Information
### Category:
- [ ] MCAD (SolidWorks)
### Type:
- [ ] Feature Enhancement
### Applicable End Item:
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software

jrsteensen, 2021-10-16:
@DG Can you add the info about the remote pot here?
@Ramjet Can you add your model here or PR it to the repo?

jrsteensen, 2021-10-16:
@dnamaster2000 Can you add the info about the remote pot here?
@Ramjet64 Can you add your model here or PR it to the repo?

jrsteensen, 2022-03-28:
Added amp to LIP, v33, located centerline, 4 inches from the back edge of the upper instrument panel mount (OH2A1A1-38). Installed with 4x 6-32 X 1.25 PHS, washer, and nylock nuts. Will need to drill clearance holes in UIP to clearance nuts on top side.

[image]

jrsteensen, 2022-03-28:
Closing with LIP v33 save.

closed 2022-03-28 by jrsteensen

Ramjet64, 2022-03-29:
That looks great.
Will be able to assist again soon.

Dai