jrsteensen/OpenHornet #372: Swap pin 2 and 3 of Q1 on DDI/AMPCD Controller

jrsteensen, 2021-10-07:
### Summary:
Swap pins 2 and 3 of Q1 on DDI/AMPCD controller. If you already have the current version of the controller, just install the through hole MOSFET with existing pin 1 to pin 1, pin 2 of the MOSFET to pin 3, and pin 3 of the MOSFET to pin 2 of the PCB.

jrsteensen, 2021-10-08:
Manufacturing files not updated. Will require manufacturing files before issue can be closed. (Gerber's, BOM, P&P file)

[JLCPCB Instructions 1](https://support.jlcpcb.com/article/149-how-to-generate-gerber-and-drill-files-in-kicad)
[JLCPCB Instructions 2](https://support.jlcpcb.com/article/84-how-to-generate-the-bom-and-centroid-file-from-kicad)

L-Walker, 2021-10-17:
Solved in commit c860438

Changed - Layout of silkscreen items to optimize readability and aesthetics.
Added - OH Logo on front silkscreen
Added - New manufacturing files (Have been tested going into JLCPCB)
Removed - Old manufacturing files.
[image]
[image]

closed 2021-10-17 by L-Walker