jrsteensen/OpenHornet #401: Painting MDF adds some thickness that may affect tolerances when assembling pieces

russellmm, 2021-11-03:
When you are assemblying the pit after painting, the thickness of the paint needs to be accounted for with the tight tolorence of the design. For example, when installing the inbord spacers, the wings fit very tight over the MDF (see picture). This can also affect the spacing of things like the lower stringers

### Summary:
See above

### Expected Results/How It Should Work:
May consider painting the parts after combing them together or leaving the connecting points unpainted until later. Could also consider adding a little tolerence to some of the pieces such as the inboard spacers. 

### Actual Results/How It Does Work:
Thick paint could cause misalignment of the ribs due to stringers pushed pieces wider than expected (by the thickness of the paint). 
You inboard spacer wings can break when trying to install them on the MDF
What I did was - slightly cut the stringers shorter than spec'd and filed down the MDF slightly around where the inboard spacers attach to the MDF ribs

### Screenshots:

[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ x] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ x] Left Console
- [ x] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-11-14:
I've spent some time thinking about this, and to not increase gaps even further when building with metric materials, I'm not going to integrate this into the model. Instead, we will address this as a flag note in the drawing.

- [ ] Add flag note to console drawings referencing rib interfacing printed parts. "Part printed to maximum material condition. Removal of .02" MAX material on MDF interfacing surfaces authorized."

jrsteensen, 2022-09-24:
Added note to #539, #541, #512

closed 2022-09-24 by jrsteensen