jrsteensen/OpenHornet #412: Clearance Issue with PCB and DDI Button Cap

dnamaster2000, 2021-11-12:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
The button caps for the DDI interfere with the Caps on the DDI PCB

### Expected Results/How It Should Work:
should be clear

### Actual Results/How It Does Work:
is not clear, fixed on my example with a file

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category
:

*Check one or more items.*

- [ X] MCAD (SolidWorks)
- [X ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-11-13:
We will make this a note to file those 4 buttons slightly in the drawing to clearance the PCB side components. My intent was the existing rim cutouts would clearance them....but apparently not.

- [ ] Add flag note to drawing, "At time of installation, file or otherwise remove material between tabs to the full thickness of tabs."

jrsteensen, 2022-09-24:
Made note on relevant issues, #505 & #537

closed 2022-09-24 by jrsteensen