jrsteensen/OpenHornet #417: Butt Kicker Amp remote

dnamaster2000, 2021-11-21:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
Here are the details for the amp remote on the X502-1

### Expected Results/How It Should Work:
N/A

### Actual Results/How It Does Work:
Remote uses an RJ-11 connector, i used an old phone cable, cut in half, the wire color in any phone cable should be the same

Pin 1 - NA
Pin 2 - GND
Pin 3 - GND
Pin 4 - Variable
Pin 5 - Positive
Pin 6 - NA

Cut cable in half (or at required length) wire Black/Red together on one side of the POT, Green to the center pin, and Yellow to the other pin.

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [X] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [X] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2022-03-28:
I was starting on this, but realized it might be a good project for a contributor. What we would need is a small printed frame that would mount a legend plate with a P081N-271F2CBR10K or P0915N-FC20BR10K pot, and a COTS RJ-11 breakout jack on the back, so we can route this from the amp to the breakout with a RJ11 bulkhead, then a RJ11 cable from the bulkhead to the remote, which will be located on the Left Console Aft interior bulkhead, fairly low to prevent interference with future console lighting. I've already started an assembly/folder in the left folder for this as the OH4A10-1.

jrsteensen, 2022-08-03:
Completed in LC v93.

closed 2022-08-03 by jrsteensen