jrsteensen/OpenHornet #418: AMPCD Board Clearance

dnamaster2000, 2021-11-28:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
There is not enough clearance in the AMPCD for the board to fit, some of the supports also line up with buttons, leaving 0 clearance between the board and the part.

### Expected Results/How It Should Work:
board should fit in to the backplate

### Actual Results/How It Does Work:
board through hole components interfere with backplate

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [x ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ x] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [x ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-11-28:
Will add greater clearance to rear bezel that the PCB seats into.

jrsteensen, 2021-11-28:
Closed with referenced commit.

commit 05a3a0b1 (2021-11-28): Update HDG-CRS Set Spacer.sldprt

Added .035 clearance and removed PCB supports.
Closes #418
closed 2021-11-28 by jrsteensen