jrsteensen/OpenHornet #421: Addition Support for RWR Control PCB

bnepethomas, 2021-12-24:
### Summary:
The RWR PCB needs additionalsupport  near the BIT FAIL button as the PCB excessively flexes when the button is pressed.  Or as a workaround add support material at the rear of the PCB (as per photo)

### Expected Results/How It Should Work:
Add a extension to the PCB along with a new heat nut in the MCAD

### Actual Results/How It Does Work:

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [X ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):

jrsteensen, 2022-07-17:
@celemourn, this one is all yours!

jrsteensen, 2022-08-07:
Added .125 THK Support Plate behind PCB and increased PCB mounting HW length to compensate. Fixed in RWR Control ASSY v4.

closed 2022-08-07 by jrsteensen