jrsteensen/OpenHornet #422: Add additional support for Bleed Air Shaft

bnepethomas, 2022-01-05:
### Summary:
In the current build of the ECS bleed air mechanism the gears occasionally bind which appears to be caused by the shaft not holding true.  Add support at the rear of the assmembly to hold shaft true

### Expected Results/How It Should Work:
Add rear support which is fractionally larger than the shaft.  As per the attached step file

### Actual Results/How It Does Work:
With the shaft no longer wobblying the gears no longer jam - have built and tested

### Screenshots:
[image]

## More Information

### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software

### Associated Filename(s):
Bleed Air Mechanism Mount.sldprt

bnepethomas, 2022-01-05:
I've got the step file if it is of interest to get the measurements

jrsteensen, 2022-04-08:
@bnepethomas Can you attach the step file to this issue please?

jrsteensen, 2022-04-08:
Completed by @erikscott128

closed 2022-04-08 by jrsteensen