jrsteensen/OpenHornet #424: COMM Panel Knob change (VOX)

Ben-F111, 2022-03-02:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
the VOX Knob on the COMM Panel is shown as a KORRY RD239, but research shows it should be a TS3750, as per the ones on the HUD Panel

### Expected Results/How It Should Work:
Change knob to TS3750,

### Actual Results/How It Does Work:
Change knob to TS3750,

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
COMM Panel

jrsteensen, 2022-03-04:
Verified across multiple sources and this change will need to happen. Thanks @Ben-F111!

Ben-F111, 2022-03-04:
[image]

looks the part in the Pit. Thanks John.

Lazarus94, 2022-03-24:
Fixed in V5 of Comm Panel (F360)

jrsteensen, 2022-03-27:
Verified change, good work @Lazarus94!

closed 2022-03-27 by jrsteensen