jrsteensen/OpenHornet #425: Dual-shaft potentiometer PTH902-030F-103B2 obsolete

erikscott128, 2022-03-11:
### Summary:
The dual concentric shaft potentiometer PTH902-030F-103B2, manufactured by BOURNS, is to be discontinued. DigiKey shows it as Obsolete. Mouser indicates it's an "End of Life" item with limited quantities to be available from a recent order until stock runs out. 

### Expected Results/How It Should Work:
The relative absence of similar, easy-to-purchase, dual concentric shaft potentiometers will likely require the design of a custom assembly utilizing 2 independent potentiometers similar to the design of the BLEED AIR and SELECT JETT knobs. 

### Actual Results/How It Does Work:
As of now, the current design should work assuming the user is still able to acquire the BOURNS potentiometer. 

### Screenshots:
Evidence here: https://www.bourns.com/docs/technical-documents/product-obsolescence-memos/Bourns_SC2118_PTH90_POM.pdf

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (Fusion360)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [X] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A5A1-1 - ASSY, ECS PANEL

erikscott128, 2022-03-29:
A new custom assembly has been designed to provide an equivalent to the PTH901 potentiometer (the PTH902 contains a switch, which is unrealistic and should originally have been a PTH901-series pot). The assembly has different installation requirements that will require modification to the backlighting PCB and backplate. Specifically, the anti-rotation hole for the obsolete pot will have to be removed, the main mounting hole enlarged, and two #2 countersunk clearance holes will have to be added. Additionally, the  connectors on the backlighting PCB may have to be moved to make space for the mating plugs. In the current position, there is only a 0.058" between the dual-pot assembly and the on-board connectors. 

[image]

jrsteensen, 2022-05-02:
@erikscott128 MCAD design is complete and tested, correct?
@L-Walker Can you update the PCB when you get the chance and commit a STEP file?

erikscott128, 2022-05-02:
MCAD is complete

L-Walker, 2022-05-03:
Sure thing.

jrsteensen, 2022-08-07:
PCB Updated in ECS v21.

closed 2022-08-07 by jrsteensen