jrsteensen/OpenHornet #427: ASSY, BULKHEAD INTERCONNECT PANEL Revision/Breakdown

Exprezz0, 2022-04-08:
### Summary:
BULKHEAD INTERCONNECT PANEL Assembly need to be broken down into three seperate assemblies.

(1) ABSIS002 - ASSY, ABSIS MEGA v1:1
(2) ABSIS003 - ASSY, ABSIS HID BUS MASTER
(3) [OH NAME] - ASSY, BULKHEAD INTERCONNECT PANEL

Currently, there isn't a anchor point for all 3 subassemblies to join together within the parent assembly

### Expected Results/How It Should Work:
Following components should be broken down into it's own subassembly and named.:
ABSIS_Bulkhead_Connector
ABSIS_AUX_Bulkhead_Connector
Right Console Bulkhead Interconnect Panel Legend Plate
CABLE, USB 2.0, Type B Female to Male
Associated Hardware

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (Fusion 360)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH5A11A1-1 - ASSY, BULKHEAD INTERCONNECT PANEL

jrsteensen, 2022-08-03:
This will be addressed when bulkhead panel is redesign along the lines of the left console.

jrsteensen, 2022-08-07:
Redesign completed in right console v95

closed 2022-08-07 by jrsteensen