jrsteensen/OpenHornet #431: 2-inch Gauges Require Cleanup/Reorganization

erikscott128, 2022-04-22:
### Summary:
Various issues remain with 2-inch dials (Airspeed, Cabin Pressure and Vertical Speed). There are common components between these (Photo-sensor, the entire back assembly, pointers, and glass face) that should be captured as such (ie, separate components that can be linked in to the respective assemblies). Various tolerances and interferences need to be worked out (This can be a separate issue if appropriate)

### Expected Results/How It Should Work:
The SINGLE STEPPER 2IN GAUGE MECHANISM (Part numbers OH2A6A1-100 and OH2A7A1-106) should be broken out as a separate common assembly and placed in either the Common folder or the Lower Instrument Panel folder. Similarly, the photo-switch (OH2A7A1-109) should be its own component as well since it is used not only in the common 2-in stepper assembly, but also in the standby altimeter and likely in other gauges as well.  

Additionally, there should be no interferences between parts. 

The Dial face should fit snuggly in assembly and bey keys to prevent rotation

### Actual Results/How It Does Work:
Currently, the Photo-switch is unique to each assembly. The common 2-inch stepper mechanism assembly is also unique to each assembly. The glass, indicator pointer and spacer should also be made common. 

There are interferences in the screws used to fasten the PCB standoffs (screws are too long). I recommend shortening the flat-head screws to 0.25" in length (currently 0.375"). The 2x4 pin header on the PCB is located directly beneath the stepper motor, and the through-hole/solder side will likely interfere with the stepper motor. 

The self-tapping screws used to retain the stepper motor are likely too small and will not thread the plastic (Assuming the holes in the 3D model are correctly sized). 

The dial face has an excessively large gap between it and the inner wall of the dial. This should be reduced, and the dial face, spacer, and (optionally) glass should be keyed to prevent rotation. 

### Screenshots:
[image]
Standoff screw interfere (Shortening flathead screw to 0.25" will fix it)

[image]
[image]
Pin headers interfere with stepper motor. 

[image]
Recommend extending PCB and moving header to this location

[image]
Large gap between dial face edge and inner wall. 

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
- [X] MCAD (Fusion360)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
- [X] Bug
- [ ] Feature Enhancement
- [X] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH2A6A1-1
OH2A6A1-100
OH2A7A1-100
OH2A7A1-102
OH2A7A1-103
OH2A7A1-105
OH2A7A1-106
OH2A7A1-100
OH2A7A1-300

jrsteensen, 2022-08-08:
| Name           | Dial Face Diameter Fixed | Spacer Keyed       | Dial Face Keyed    |
|----------------|--------------------------|--------------------|--------------------|
| CABIN PRESS    | :heavy_check_mark:       | :white_check_mark: | :white_check_mark: |
| STBY ALTIMETER | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| STBY AIRSPEED  | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| STBY VVI       | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| RALT           | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| HYD PRESS      | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| BATT           | :white_check_mark:       | :white_check_mark: | :white_check_mark: |
| BRAKE PRESS    | :white_check_mark:       | :white_check_mark: | :white_check_mark: |

jrsteensen, 2022-08-08:
All 2 inch gauges complete. No changes to RALT, HYD PRESS, BATT or BRAKE PRESS due to other constraints.

closed 2022-08-08 by jrsteensen