jrsteensen/OpenHornet #434: Left Lower Instrument Panel Printability Concerns

erikscott128, 2022-04-23:
### Summary:
Left lower instrument panel has some printability issues that could be resolved

### Expected Results/How It Should Work:
Left lower instrument panel should be printable without excessive use of support material

### Actual Results/How It Does Work:
Significant overhangs make printing difficult. Recommend separating these flanges into separate pieces that are screwed in, similar to what is done on the right side. 

### Screenshots:
[image]
Highlighted faces show the overhangs that should be separated

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH2A2A1-10

Ben-F111, 2022-04-23:
On our pit. These have been replaced with screw on supports. After screen is installed.

erikscott128, 2022-04-23:
@Ben-F111 Can you provide pictures of your solution?

jrsteensen, 2022-07-16:
Agreed. Can you implement it @erikscott128?

erikscott128, 2022-07-17:
Complete: 
[image]

closed 2022-07-17 by erikscott128