jrsteensen/OpenHornet #453: Additional Information in relation to Issue #438 - Assembly of Standby Altimeter Vid29 Stepper and J1 Header Pin

aegis53, 2022-08-05:
Reference - Github issue #438 - Assembly of Standby Altimeter Vid29 Stepper and J1 Header Pin

### Summary:
I have continued building this module and now mounted the Zero-Detect Module onto the assembly.

Everything regarding this module stems back the fact that J1 Header interferes with the flush mounting of the Stepper against the PCB.  I have determined that my original workaround to raise the Stepper ultimately is not a solution and that the PCB will most certainly need a redesign at the very least. (Move J1 Header pins to the lower edge of the PCB.)
In raising the stepper I have encountered another issue right at the end. This extra height of the stepper now reduces the clearance in relation to the Zero-wheel rotation through the IR Detector on the Zero_Detect module. I'll try to capture a picture of this and upload it shortly.
[image]

A couple of other observations here, I have included possible suggestions in (Brackets)...
1. I'm not sure how the actual zero_detect Wheel is fixed to the Stepper shaft, (assume Hot Glue perhaps?)
2. How does the Zero_Detect module actually connect to the Zero_detect header Pins, because there is not enough clearance to actually use a female 3 pin header plug within the assembly. In which case a VERY!! short cable run would need to be soldered directly to the Zero_detect module and the Zero_detect header pins internally to the  altimeter and thus possibly fouling the zero_detect wheels rotation. Fabricating a 1cm cable in length is quite difficult. (Try and integrate the Zero Detect module or sensor directly into the PCB, noting that the IR module needs to be mounted at 90 degrees to allow the wheel to rotate this could be tricky.......??? 

Further suggestions/Possible Options.
1. Obtain/use steppers that place the connecting pins on the face of the stepper and not the rear, this would essentially mount the stepper on the other side of the PCB to the position its now in. With only the Shaft protruding through a hole. This removes the clearance issues, but needs a whole of PCB adjustment.

Current solution as built.
After giving this some thought using the parts as is, my planned option is to use standoffs of between 12-15mm in length, this should solve all clearance issues, but will also require longer header socket leads for the OLED displays and its possible the Stepper shaft and ultimately the Altimeter needle will need a small extension using Plastic tubing. 
The extra clearance would assist in providing a support for the OLED's using plastic strip.

### Expected Results/How It Should Work:
In summary these are a list of issues I see with the module as I have built it.

1. The Stepper should mount flush to the PCB and not interfere with the J1 header pins protruding through the PCB. (PCB Redesign.)
2. OLED displays need sturdier mounting to prevent movement within the module. (Perhaps modify the 3D printed mount??)

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [X] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
Altimeter.sldasm

jrsteensen, 2022-08-05:
PCB Redesign already completed this morning.

OLEDS will have to be mounted in hot glue. The mating face towards the stepper mount has additional components on the OLED, so a hard surface for the support cannot be implemented well, and the wheel is immediately behind the OLED.

Zero wheel just gets a dollop of hot glue to fasten on shaft roughly centered between arms of IR sensor.

Zero sensor is connected by wires soldered into PCB, then route cables to controller PCB, securing the wires as required as to not mechanically interfere with the movement of the zero wheel.

jrsteensen, 2022-08-08:
No further action indicated at this time than above. Closing for now, but please let me know if we need to look at this again.

closed 2022-08-08 by jrsteensen