jrsteensen/OpenHornet #455: ECM Button Holder needs Tabs

Ben-F111, 2022-08-05:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
ECM Button Holder needs Tabs

### Expected Results/How It Should Work:
The button for the ECM JETT requires a way of retaining the button housing 

### Actual Results/How It Does Work:
The current .55 housing does not allow for a retention method 

### Screenshots:
[image]

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
OH2A5A1-105 - .55 SQ INDICATOR BASE

Ben-F111, 2022-08-05:
[image]

completed unit

[image]

jrsteensen, 2022-08-08:
Fixed in ECM Panel v4

closed 2022-08-08 by jrsteensen