jrsteensen/OpenHornet #456: Add 2 Screw holes to KY58 Aft Legend and Light Plates

russellmm, 2022-08-07:
When creating the KY58 panels, I noticed there was no screw holes to connect the Aft Legend and Light plates to the Backplate. I assume the intent is to weld/glue them but was curious if this was how the aircraft was done. In researching, I was able to find one image of the KY58 on Wikipedia at a Crypto muesum. 

### Summary:
(Please summarize your issue here. Be specific.)

### Expected Results/How It Should Work:
(Explain how you think it should work here. Be very specific.)

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*

[image]

https://en.wikipedia.org/wiki/KY-58#/media/File:KY-58_VINSON,_airborne_or_shipborne_-_National_Cryptologic_Museum_-_DSC08023.JPG

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

russellmm, 2022-08-07:
This link says its for battery install and they have a few images without the screws. So maybe its not really needed
http://www.jproc.ca/crypto/ky58.html

Ben-F111, 2022-08-07:
Will
Send 
Pic
Later. But yes. Glue.

closed 2024-02-25 by jrsteensen