jrsteensen/OpenHornet #462: Landing Gear Pushbutton Switch Issue

Stang51, 2022-08-12:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(The Warning Tone Silence Pushbutton "Barrel" is too short - Not enough extension after passing through the Legend Plate. )

### Expected Results/How It Should Work:
(The Warning Silence Pushbutton should extend through the Legend Plate by a minimum of .125")

### Actual Results/How It Does Work:
(Currently with the "Original Solidworks Design", the Warning Silence Pushbutton extends through the Legend Plate by less that 0.0625".   This makes it very hard to actually press the button far enough in, so that the contacts in the switch touch.)

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]

[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X ] Bug
- [ X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
Landing Gear Lever Panel.sldasm

jrsteensen, 2022-08-12:
This was addressed at a previous time. (Replaced with KM1101RR02QE switch)
Closing with no action required.

jrsteensen, 2022-08-12:
I may have been a bit too hasty on closing this. Erik, can you look at this as part of #441?

jrsteensen, 2022-08-12:
Nope, already addressed with change of switch and custom switch cap. Closing this with no action.

closed 2022-08-12 by jrsteensen