jrsteensen/OpenHornet #463: Panel Backlighting: Option B

erikscott128, 2022-08-15:
### Summary:
The existing backlighting method utilizes a multi-layer method that can be accomplished entirely using a laser cutter (and a manufactured PCB). While it provides good results, there are several issues associated with this design. A method more in-line with what is done on actual panels provides several significant benefits at the expense of slightly more difficult fabrication. 

### Current Method ("Option A"):
#### Overview:
The current backlighting method divides the panel into 4 layers. The top-most layer, referred to as the "legend plate" is a laser-engravable material consisting of a thin black upper surface and translucent white body. Panel legends and insignia are laser-engraved on the upper surface such that the white material is revealed. The second layer, referred to as the "light plate", is a translucent white acrylic layer used to diffuse light within the panel. The third layer is an opaque black acrylic layer, referred to as the "backplate", to which most switches, buttons and other panel components mount. Below that is the backlighting PCB, which contains connectors and a capacitor on its lower side, and addressable RGB SMD LEDs (red-green-blue, surface-mounted light-emitting diodes) on its upper surface. The LEDs are recessed in cutouts within the backplate. They shine up into the translucent light plate and the light ultimately escapes through the engravings in the upper surface of the legend plate, or is absorbed within the light plate or legend plate. The LEDs are positioned directly below the engraved labels, and are packed densely to provide even light distribution and avoid so-called "hot-spots" (distinctly bright areas) of the appearance of the backlighting. The individually addressable LEDs provide the ability to dim or brighten specific LEDs to further reduce hot-spots. Because of the number of LEDs and the relative density of the engraved areas on the legend plate on most panels, the number and size of the recesses required on the backplate become fairly extensive, resulting in a "swiss-cheesed" look of the backplate, which reduces the strength of this layer. This issue is largely mitigated by the strength provided by the other layers. The backplate must also be thick enough to accommodate the LEDs, usually necessitating 1/8" or 3mm thick acrylic. This is in addition to the thickness of the backlighting PCB, which is at least 1mm for reasonably-priced custom boards. The entire stack-up is held together with No. 6-32 screws. The edges of the legend plate and light plate must be manually painted black to prevent light-leakage. The light and legend plates can be optionally fused together to make a single panel similar to the light plates found in actual aircraft (see the real-life implementation section for details). 

#### Benefits:
This method was devised by @L-Walker and has several benefits. Chief among these is the relative ease of construction, provided the builder has the appropriate materials and access to a laser cutter/engraver. The PCBs can be manufactured relatively cheaply from PCB manufacturing houses like JLCPCB with SMDs and connectors already soldered to the board. With each layer fabricated, the builder need only mount the switched to the backplate and backlighting PCBs and assemble the rest of the sandwich using No. 6 screws after painting the edges of the light and legend plates. 

The individually addressable LEDs provide the ability to tune the brightness of specific LEDs to help eliminate hot-spots. Additionally the ability to flash the legends associated with a specific switch can help utilize the sim-pit as a training device, or help identify a switch during the setup phase. 

#### Disadvantages: 
The necessity to use a 1/8" or 3mm backplate, plus the minimum 1mm backlighting PCB thickness results in a ~4mm thick panel for switches to mount to. The threaded bushing on many switches does not allow for this with sufficient thread engagement of the associated mounting nuts. Sometimes, even if sufficient thread does exist, the shafts of potentiometers, encoders, and rotary switches does not protrude high enough above the legend plate and into the associated knob to permit the set screws to grip the shaft. This limits the selection of switches available to use in the designs of panels and sometimes required the modification of knobs to re-position set-screws. 

The requirement that the LEDs be recessed in the backplate requires that extensive amounts of the backplate be removed, resulting in the afore-mentioned "swiss cheesed look. This can potentially make the backplate relatively fragile and prone to breakage during assembly. Once assembled, however, this fragility is largely mitigated by the structure provided by the solid PCB from below and solid light and legend plates above. 

The large amounts of space required by the LEDs, and the fact that the PCB is located below the light plate limits mounting options for switches, stand-offs, and more complex assemblies on the underside of the panel. Since screws cannot penetrate through areas located where LEDs are, and LEDs are located wherever legends are placed on the upper surface, mounting positions are very limited, especially on compact, non-standard panels such as the LDG GEAR panel. 

Because the light-plate is translucent rather than transparent, light does not travel very far within the panel before being absorbed. This requires an extensive number of LEDs. This makes PCBs more expensive, presents potential power consumption issues, and exacerbates the issues presented above regarding backplate swiss-cheesing and mounting space. 

### Real-life implementation:
#### Overview:
The specs for real-world panels can be found in MS25212. In short, they consist of an upper plastic "light plate" which features the engraved legends, and a lower "mounting plate" to which all switches and other panel assemblies are mounted. The light plate is attached to the mount plate via No. 6-sized screws and nut-plates attached to the mounting plate. The finished light plate is to be 0.240±0.023 inch thick as per MIL-DTL-7788. The mounting plate is to be 0.064±0.003 inch thick as per MS25212. 

Backlit panels within real-world aircraft in the US militaries come in several varieties as the years have gone on. Externally, all adhere to roughly the same physical specs, and internally, while the exact methods have ben changed refined over the years, all operate on the same working principle: Light is emitted within a transparent acrylic "light plate" which has been painted white and then black such that it has a white internal surface and black external surface. The light bounces and scatters off of the internal surface until it can escape through an engraving in the upper surface of the panel. The light plate connects electrically to the rest of the backlighting circuit via a plug connector on the bottom surface which seats in a receptacle mounted to the mount plate. The location of this connector is denoted on the *top* surface of the panel by an un-illuminated white cross or + symbol. 

The lights contained within the light plate are generally placed away from engraved legends (ie, not directly below engraved areas). The quantity of embedded lamps is specified to be no more than 10, though it is evident that 12 were used for the ECS panel. 

There are several types of backlit panels, the details of which are described in MIL-DTL-7788. Of note are Type IV, Type V, and Type VII panels. 

#### Type IV - One piece integrally wired incandescent panels
These panels' light plates contain incandescent bulbs (MS90451 or MS90452) embedded within the panels. See the pictures section for an example of this type of light plate from a real F/A-18 ECS panel. The locations of the bulbs and the wire traces are depicted on the back in white paint. The backlighting circuit within the panel connects to a receptacle in the mounting plat via a MS90335-7 connector. 

#### Type V - Printed circuit board panels
These panels have a recess in the rear surface of the panels for a removable circuit-board to which incandescent lamps (MS90451 or MR90452) are soldered. The circuit board is attached to the light plate via screws (No. 2 size) and the bottom surfaces of the plastic panel and circuit board should be flush. The electrical connection is made via an MS90335-8 connector. 

#### Type VII - Light emitting diode (LED) panels
[... cut ...]
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [X] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

L-Walker, 2022-08-15:
Very detailed breakdown. I'm a straight shooter and am just going to say,
my boards are there, they are almost all done. The RWR and Defog lever are
the only boards not yet with known working models. I've spent the cash
getting them working.

If you do want to go the other route that is totally fine but either way I
can no longer contribute in any way due to an increase in workload running
my company.

Luke

tekadept, 2022-09-07:
A short little pdf that describe a little on the process of real panels constructions I found in my travels
https://www.primeproductsinc.com/wp-content/uploads/Illuminated_Components_Design_Guide-final.pdf