jrsteensen/OpenHornet #465: BRAKE PRESSURE INDICATOR Shroud overlaps with Rear Spacer

russellmm, 2022-08-19:
### Summary:
OH4A3A3-1 - ASSY, BRAKE PRESSURE INDICATOR 
The OH4A3A3-12 - BRAKE PRESSURE INDICATOR SHROUD seems to be too long and it's overlapping with the OH4A3A3-16 - BRAKE PRESSURE INDICATOR REAR BEZEL SPACER. I don't think it is a simple case of moving the Shroud as then the Indicator bezel face would sit way too far into the gauge. 

### Expected Results/How It Should Work:
It looks like the mating of the two parts needs to be corrected to allow for the proper depth and proper mating.

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.

[image]

*Check one or more items.*
- [ X] MCAD (SolidWorks)

### Type:
*Check one item.*
- [ X] Bug

### Applicable End Item:
*Check one item.*
- [ X] Left Console

### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2022-08-19:
Huh. Oops. Will fix.

jrsteensen, 2022-08-19:
Completed in LC v95.

closed 2022-08-19 by jrsteensen