jrsteensen/OpenHornet #466: Decide if we want to make all hubs USB 3.

jrsteensen, 2022-08-19:
Decide if we want to make all hubs USB 3. Can be facilitated with minor wiring changes and no changes to bulkhead connections, just increases cabling and hub cost somewhat.

russellmm, 2022-08-20:
Yes please. USB3 likely to be available longer than USB2

jrsteensen, 2022-08-27:
Yeah, might as well future proof it. We will implement this.

- [ ] Replace USB 2.0 HUB in UIP
- [ ] Replace USB 2.0 HUB in RC
- [ ] Replace USB 2.0 HUB in LC
- [ ] Update LC Interconnect
- [ ] Update RC Interconnect
- [ ] Update LIP Interconnect
- [ ] Update RC>LIP Bulkhead with USB 3 Connector
- [ ] Update LC>LIP Bulkhead with USB 3 Connector
- [ ] Update LIP>LC Bulkhead with USB 3 Connector
- [ ] Update LIP>RC Bulkhead with USB 3 Connector
- [ ] Update UIP>LIP Bulkhead with USB 3 Connector

commit bbd6d873 (2022-09-10): Made all interfaces USB 3.0 compatible.

All interfaces USB 3.0 compatible.
Closes #466
closed 2022-09-10 by jrsteensen