jrsteensen/OpenHornet #471: VVI Bezel - currently single body

Ben-F111, 2022-08-30:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(VVI Bezel Printing)

### Expected Results/How It Should Work:
(Where posible its prefered to print 'deisplay" parts Face down on the printer bed)

### Actual Results/How It Does Work:
(The new VVI Bezel is a single part with the ring protrusion attached to the body, this is not allowing the Bezel visible face(s) to be printed flat on the bed, thus providing a superior print finish with limited post processing, 

suggest the VVI Bezel and Ring are 2 independent Bodies (or 2 STLs in the final) that can be printed face down then glued prior to printing

Printing the Standby Gauge Bezels Face down also removes the needs for any support material )

### Screenshots:
I have split and will upload resulst later today

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

russellmm, 2022-08-30:
could just split it and a dab of glue and its good to go
[image]

Ben-F111, 2022-08-30:
yes, correct, noting there a some ppl that dont know how to do that, the request is to relased STL's to be split already 

I would split if from the top of the BEZEL face, as a smaller ring, not the entire section including the Dial Ring

Will have pics later today

Ben-F111, 2022-08-31:
[image]

top Bezel printed as 2 parts. 
[image]

Ben-F111, 2022-08-31:
[image]

jrsteensen, 2022-09-10:
[image]
[image]

Completed in VVI v7.

closed 2022-09-10 by jrsteensen

russellmm, 2022-09-10:
looks great!