jrsteensen/OpenHornet #475: Redesign Rudder Trim Knob Mechanism

jrsteensen, 2022-09-08:
Existing Rudder Trim Mechanism does not work. (Servo is underpowered.)

Evaluate Rudder Trim Knob Mechanism and redesign as appropriate.

Ben-F111, 2022-09-10:
I can update to the servo I used. It works great. ?

jrsteensen, 2022-09-10:
Yes please! I would rather do that for Beta1, and we can do a full redesign at a later time if required.

Ben-F111, 2022-09-17:
Update F3D  and servo type provided to @jrsteensen for evaluation as used in my pit.

jrsteensen, 2022-09-20:
Thanks for that @Ben-F111. @erikscott128 came up with a motorized pot (pot with an integrated DC motor) that I'm excited to try out. I archived all your stuff for future reference if required. 

FCS panel is completed, just waiting to close until I redesign the interfering APU ABSIS PCB mount in the console assy.

jrsteensen, 2022-09-23:
Complete in LC v123. So so close! One more issue!

closed 2022-09-23 by jrsteensen