jrsteensen/OpenHornet #482: Create a 3D Printed File Standard

jrsteensen, 2022-09-12:


jrsteensen, 2023-04-01:
Will add additional requirements if anyone comes up with any extra that make sense.

closed 2023-04-01 by jrsteensen

kbastronomics, 2023-04-01:
I'm sure you have your reasons but shouldn't what the part is be part of the naming.

so say
FDM_OH5A5A1-18_-_v22.stl
it was instead

FDM_OH5A5A1-18 - ECS PANEL ABSIS ADAPTER PLATE_-_v22.stl
now I as the one working on this don't have to refer back to the fusion model to figure out what OH5A5A1-18 was

So your naming would be virtually the same just
FDM_<PNAME>_<REV>.<VERSION # OF MODEL>.stl
PNAME would pull from the properties page

jrsteensen, 2023-04-01:
You know, thats a fair point. The intent was to force the user to reference the drawing by only listing the PN. 

My issue with changing it at this point is we have a bunch of drawings we would have to update the standard notes on for beta1. That said, instead of listing it on EVERY drawing, we should have listed the naming convention in the OH-SPEC-001.

The other issue I ran into is some file names with the nomenclature get too long to be recognized on the printer.

kbastronomics, 2023-04-01:
100% of the time I would still print straight from fusion as STL files are old and not recommended for use any more so I create  3mf files.   but that's my personal preference.

jrsteensen, 2023-04-01:
That kinda goes against having a release package. The idea of the release is "this all works together right now" whereas the fusion will be the "in-progress" source that may or may not work.

kbastronomics, 2023-04-01:
just wondering if you should go with 3mf or obj files maybe.
maybe not this release but something to consider