jrsteensen/OpenHornet #496: Create Drawing: OH1A6 - ASSY, PANEL, SPIN RECOVERY

jrsteensen, 2022-09-14:
Create drawing OH1A6 - ASSY, PANEL, SPIN RECOVERY.

jrsteensen, 2023-04-02:
IN CHECK

[OH1A6-1 - ASSY, PANEL, SPIN RECOVERY_-_v1.pdf](https://github.com/jrsteensen/OpenHornet/files/11131139/OH1A6-1.-.ASSY.PANEL.SPIN.RECOVERY_-_v1.pdf)

kbastronomics, 2023-04-02:
How does one attach the Spin Indicator,   that's not clear or I'm blind and am missing it

jrsteensen, 2023-04-02:
No, valid catch - it gets bonded to the backplate. Unfortunately its too thin to be able to have a mounting boss easily attached, since the full clearance is required from the rear to insert the legend stack.

- [ ] Add note detailing bonding of SPIN indicator.

jrsteensen, 2023-04-02:
Added bonding note

[OH1A6-1 - ASSY, PANEL, SPIN RECOVERY_-_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11131337/OH1A6-1.-.ASSY.PANEL.SPIN.RECOVERY_-_v2.pdf)

jrsteensen, 2023-04-02:
Self-checked. Added to RC Release package and closed.

closed 2023-04-02 by jrsteensen