jrsteensen/OpenHornet #497: Create Drawing: OH1A7 - ASSY, PANEL, HUD

jrsteensen, 2022-09-14:
Create drawing OH1A7 - ASSY, PANEL, HUD.

jrsteensen, 2023-04-02:
ON HOLD - BL PCB interference issues with nuts and poorly placed LEDs.

Exprezz0, 2023-04-22:
Drawing ready to be completed.  PCB and Fusion 360 Models have been updated per commit: cbd3a8d2b94fde9d1af5b288b04dab8aa95fb0d0

jrsteensen, 2023-04-22:
MCAD Changes TODO for me:

- [ ] Reorganize panel from -100 to -1
- [ ] Move inserts to -10
- [ ] Evaluate BLK LVL and AOA knobs - not sure we should have the white circle visible.

jrsteensen, 2023-04-22:
Drawing completed, self-checked and added to RC Release Package

[OH1A7-1 - ASSY, PANEL, HUD_-_v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11302319/OH1A7-1.-.ASSY.PANEL.HUD_-_v4.pdf)

closed 2023-04-22 by jrsteensen