jrsteensen/OpenHornet #509: Create Drawing: OH2A7A1 - ASSY, STANDBY INSTRUMENT MODULE

jrsteensen, 2022-09-14:
Create drawing OH2A7A1 - ASSY, STANDBY INSTRUMENT MODULE.

Ben-F111, 2022-10-09:
Build Note, Zero Sensor within the gauges

Add to the drawing
###Zero Sensor Tip Must be painted as per F360 - with at least 3 coats of Spray-paint. ###

the Photo Sensor we are finding is penetrating Resin Prints, heat shrink

Painting the tip will stop the beam

jrsteensen, 2023-04-16:
IN CHECK

[OH2A1A1-1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE_-_v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11243043/OH2A1A1-1.-.ASSY.LOWER.INSTRUMENT.PANEL.STRUCTURE_-_v4.pdf)

jrsteensen, 2023-04-22:
Uploaded wrong drawing.

[OH2A7A1-1 - ASSY, STANDBY INSTRUMENT MODULE_-_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11301657/OH2A7A1-1.-.ASSY.STANDBY.INSTRUMENT.MODULE_-_v2.pdf)

Self-Checked and added to RC Release Package

closed 2023-04-22 by jrsteensen