jrsteensen/OpenHornet #516: Create Drawing: OH4A3A3 - ASSY, BRAKE PRESS GAUGE

jrsteensen, 2022-09-14:
Create drawing OH4A3A3 - ASSY, BRAKE PRESS GAUGE.

jrsteensen, 2022-09-25:
IN CHECK

[OH4A3A3-1 - ASSY, BRAKE PRESSURE INDICATOR_-_v3.pdf](https://github.com/jrsteensen/OpenHornet/files/9641088/OH4A3A3-1.-.ASSY.BRAKE.PRESSURE.INDICATOR_-_v3.pdf)

jrsteensen, 2022-09-25:
[OH4A3A3-1 - ASSY, BRAKE PRESSURE INDICATOR_-_v5.pdf](https://github.com/jrsteensen/OpenHornet/files/9641135/OH4A3A3-1.-.ASSY.BRAKE.PRESSURE.INDICATOR_-_v5.pdf)

SLIGHTLY UPDATED

Lazarus94, 2023-04-01:
IN CHECK

[OH4A3A3-1 - ASSY, BRAKE PRESSURE INDICATOR v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11129204/OH4A3A3-1.-.ASSY.BRAKE.PRESSURE.INDICATOR.v5.pdf)

jrsteensen, 2023-04-01:
Added to RC Release Package and closed.

closed 2023-04-01 by jrsteensen