jrsteensen/OpenHornet #521: Create Drawing: OH4A5A2 - ASSY, APU/ENG CRANK PANEL

jrsteensen, 2022-09-14:
Create drawing OH4A5A2 - ASSY, APU/ENG CRANK PANEL.

jrsteensen, 2022-09-25:
IN CHECK

[OH4A5A2-1 - ASSY, APU ENG CRANK PANEL_-_v1.pdf](https://github.com/jrsteensen/OpenHornet/files/9641449/OH4A5A2-1.-.ASSY.APU.ENG.CRANK.PANEL_-_v1.pdf)

jrsteensen, 2022-09-25:
ADDED MISSING PO NOTE

[OH4A5A2-1 - ASSY, APU ENG CRANK PANEL_-_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/9641454/OH4A5A2-1.-.ASSY.APU.ENG.CRANK.PANEL_-_v2.pdf)

Lazarus94, 2023-04-01:
[OH4A5A2-1 - ASSY, APU ENG CRANK PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11129096/OH4A5A2-1.-.ASSY.APU.ENG.CRANK.PANEL.v3.pdf)

IN CHECK

jrsteensen, 2023-04-01:
Added to RC Release Package and closed.

jrsteensen, 2023-04-13:
Added to release package

[OH4A5A2-1 - ASSY, APU ENG CRANK PANEL_-_ v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11227220/OH4A5A2-1.-.ASSY.APU.ENG.CRANK.PANEL_-_.v6.pdf)

closed 2023-04-13 by jrsteensen