jrsteensen/OpenHornet #524: Create Drawing: OH4A7A2 - ASSY, OBOGS PANEL

jrsteensen, 2022-09-14:
Create drawing OH4A7A2 - ASSY, OBOGS PANEL.

Lazarus94, 2023-03-30:
Move to _In Progress_

Lazarus94, 2023-03-31:
[OH4A7A2-1 - ASSY, OBOGS PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11123058/OH4A7A2-1.-.ASSY.OBOGS.PANEL.v3.pdf)

Lazarus94, 2023-03-31:
[OH4A7A2-1 - ASSY, OBOGS PANEL v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11123496/OH4A7A2-1.-.ASSY.OBOGS.PANEL.v6.pdf)

Correction use this one. Added trail lines

jrsteensen, 2023-03-31:
### Check Comments

- [ ] Add Resin Note to 2.3
- [ ] FN 17 not ballooned
- [ ] FN 20 not ballooned
- [ ] Add `INITIAL RELEASE` to rev history description field.

[image]

Lazarus94, 2023-03-31:
[OH4A7A2-1 - ASSY, OBOGS PANEL v7.pdf](https://github.com/jrsteensen/OpenHornet/files/11124341/OH4A7A2-1.-.ASSY.OBOGS.PANEL.v7.pdf)

IN BACKCHECK

jrsteensen, 2023-03-31:
Saved to RC release package, and marked complete.

closed 2023-03-31 by jrsteensen