jrsteensen/OpenHornet #525: Create Drawing: OH4A7A3 - ASSY, MC/HYD PANEL

jrsteensen, 2022-09-14:
Create drawing OH4A7A3 - ASSY, MC/HYD PANEL.

jrsteensen, 2022-09-25:
IN CHECK

[OH4A7A3-1 - ASSY, MC HYD ISOL PANEL_-_v1.pdf](https://github.com/jrsteensen/OpenHornet/files/9641483/OH4A7A3-1.-.ASSY.MC.HYD.ISOL.PANEL_-_v1.pdf)

jrsteensen, 2023-03-27:
Need to update HYD/ISOL switch to latest.

Lazarus94, 2023-03-31:
[OH4A7A3-1 - ASSY, MC HYD ISOL PANEL v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11125296/OH4A7A3-1.-.ASSY.MC.HYD.ISOL.PANEL.v2.pdf)

IN CHECK

jrsteensen, 2023-03-31:
[OH4A7A3-1 - ASSY, MC HYD ISOL PANEL_-_v2-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11125363/OH4A7A3-1.-.ASSY.MC.HYD.ISOL.PANEL_-_v2-CHK1.pdf)

Lazarus94, 2023-04-01:
IN BACKCHECK

[OH4A7A3-1 - ASSY, MC HYD ISOL PANEL v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11128980/OH4A7A3-1.-.ASSY.MC.HYD.ISOL.PANEL.v4.pdf)

Part numbering seems to have resolved itself. The mm after 18MM is stuck in capitals due to the table properties.

jrsteensen, 2023-04-01:
Just a couple tiny things:
[image]
[image]

jrsteensen, 2023-04-02:
Incorporated backcheck comments and added to RC release package and closed.

[OH4A7A3-1 - ASSY, MC HYD ISOL PANEL_-_v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11132963/OH4A7A3-1.-.ASSY.MC.HYD.ISOL.PANEL_-_v5.pdf)

closed 2023-04-02 by jrsteensen