jrsteensen/OpenHornet #527: Create Drawing: OH4A8A1 - ASSY, LEFT CIRCUIT BREAKERS

jrsteensen, 2022-09-14:
Create drawing OH4A8A1 - ASSY, LEFT CIRCUIT BREAKERS.

Lazarus94, 2023-03-31:
IN PROGRESS

Lazarus94, 2023-03-31:
[OH4A8A1-1 - ASSY, LEFT CIRCUIT BREAKERS v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11125173/OH4A8A1-1.-.ASSY.LEFT.CIRCUIT.BREAKERS.v2.pdf)

IN BACKCHECK

jrsteensen, 2023-03-31:
IN BACKCHECK

[OH4A8A1-1 - ASSY, LEFT CIRCUIT BREAKERS_-_v2-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11125215/OH4A8A1-1.-.ASSY.LEFT.CIRCUIT.BREAKERS_-_v2-CHK1.pdf)

Lazarus94, 2023-03-31:
[OH4A8A1-1 - ASSY, LEFT CIRCUIT BREAKERS Drawing v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11127001/OH4A8A1-1.-.ASSY.LEFT.CIRCUIT.BREAKERS.Drawing.v3.pdf)

jrsteensen, 2023-03-31:
Good as-is. Added to RC release package and closed.

closed 2023-03-31 by jrsteensen