jrsteensen/OpenHornet #528: Create Drawing: OH4A8A2 - ASSY, DISPENSE BUTTON

jrsteensen, 2022-09-14:
Create drawing OH4A8A2 - ASSY, DISPENSE BUTTON.

Lazarus94, 2023-03-31:
IN PROGRESS

Lazarus94, 2023-03-31:
[OH4A8A2-1 - ASSY, DISPENSE BUTTON v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11124263/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON.v3.pdf)

IN BACKCHECK

Lazarus94, 2023-03-31:
[OH4A8A2-1 - ASSY, DISPENSE BUTTON  v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11124310/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON.v4.pdf)

Added trail lines. IN BACKCHECK

jrsteensen, 2023-03-31:
IN BACKCHECK

[OH4A8A2-1 - ASSY, DISPENSE BUTTON_-_v4-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11125165/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON_-_v4-CHK1.pdf)

Lazarus94, 2023-03-31:
[OH4A8A2-1 - ASSY, DISPENSE BUTTON v9.pdf](https://github.com/jrsteensen/OpenHornet/files/11126972/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON.v9.pdf)

IN CHECK

jrsteensen, 2023-03-31:
Verify all previous redlines not incorporated are incorporated plus:

[image]
[image]

Lazarus94, 2023-04-01:
IN BACKCHECK

[OH4A8A2-1 - ASSY, DISPENSE BUTTON v11.pdf](https://github.com/jrsteensen/OpenHornet/files/11129047/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON.v11.pdf)

I can delete the additional view of the Button Head if it's too much.

Lazarus94, 2023-04-01:
IN BACKCHECK

[OH4A8A2-1 - ASSY, DISPENSE BUTTON v12.pdf](https://github.com/jrsteensen/OpenHornet/files/11129353/OH4A8A2-1.-.ASSY.DISPENSE.BUTTON.v12.pdf)

Slightly corrected version.

jrsteensen, 2023-04-01:
Added to RC Release package and closed.

closed 2023-04-01 by jrsteensen