jrsteensen/OpenHornet #529: Create Drawing: OH4A8A3 - ASSY, NUC ENABLE SWITCH

jrsteensen, 2022-09-14:
Create drawing OH4A8A3 - ASSY, NUC ENABLE SWITCH.

Lazarus94, 2023-03-28:
[OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_Drawing_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11093011/OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_Drawing_v2.pdf)

Questions:

1 Is Rowmark included as a flat part?
2. What is meant by REV?
3.  Did I explode the model correctly or should I have done it differently?

Exprezz0, 2023-03-28:
I noticed the placard, too.  Says rowmark, so you'd think it would be like the rest of the panels, but the model shows it as a sticker/printed as part of the PETG.  Same thing as the L/R circuit breaker boxes

REV = Revision.

If you can move the placard (which I don't remember if you can or not), then I'd separate it from the box a little more.

jrsteensen, 2023-03-28:
IN BACKCHECK

[OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_-_v2-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11095258/OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_-_v2-CHK1.pdf)

jrsteensen, 2023-03-29:
Use this one, missed one little thing (The "LEFT CONSOLE" in the title block)

[OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_-_v2-CHK2.pdf](https://github.com/jrsteensen/OpenHornet/files/11095323/OH4A8A3-1_-_ASSY_NUC_WEAP_SWITCH_-_v2-CHK2.pdf)

Lazarus94, 2023-03-29:
Thanks John, that helps a lot. I’ll send the corrected version this afternoon.

Lazarus94, 2023-03-30:
[OH4A3A1-1.-.ASSY.SELECT.JETT.PANEL_-_v5-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11114280/OH4A3A1-1.-.ASSY.SELECT.JETT.PANEL_-_v5-CHK1.pdf)

Lazarus94, 2023-03-30:
[OH4A8A3-1 - ASSY, NUC WEAP SWITCH Drawing v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11115067/OH4A8A3-1.-.ASSY.NUC.WEAP.SWITCH.Drawing.v5.pdf)

jrsteensen, 2023-03-30:
Just as a minor note: Drop the word "Drawing" from the PDF. The PDFs should be named per the redlines in -CHK2. `<DWG #> - <DWG TITLE>_<REV>_v<VER>.pdf`

Thanks Laz! Saved to the release candidate package.

closed 2023-03-30 by jrsteensen