jrsteensen/OpenHornet #530: Create Drawing: OH10A1A1 - ASSY, SIM PWR PANEL

jrsteensen, 2022-09-14:
Create drawing OH10A1A1 - ASSY, SIM PWR PANEL.

Lazarus94, 2023-03-31:
Move to _In Progress_

Lazarus94, 2023-03-31:
[OH10A1A1-1 - ASSY, SIM PWR PANEL v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11123492/OH10A1A1-1.-.ASSY.SIM.PWR.PANEL.v4.pdf)

Lazarus94, 2023-03-31:
[OH10A1A1-1 - ASSY, SIM PWR PANEL v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11123681/OH10A1A1-1.-.ASSY.SIM.PWR.PANEL.v6.pdf)

Done. Thank you.

jrsteensen, 2023-03-31:
We are entering the realm of nitpicky, but:

[image]

Lazarus94, 2023-03-31:
[OH10A1A1-1 - ASSY, SIM PWR PANEL v7.pdf](https://github.com/jrsteensen/OpenHornet/files/11124283/OH10A1A1-1.-.ASSY.SIM.PWR.PANEL.v7.pdf)

IN BACKCHECK

jrsteensen, 2023-03-31:
Drawing completed and added to RC release package. Great work, thank you!

closed 2023-03-31 by jrsteensen