jrsteensen/OpenHornet #532: Create Drawing: OH4A6A1 - ASSY, FCS PANEL

jrsteensen, 2022-09-14:
Create drawing OH4A6A1 - ASSY, FCS PANEL.

Lazarus94, 2023-04-01:
IN PROGRESS

Lazarus94, 2023-04-13:
IN CHECK

[OH4A6A1-1 - ASSY, FCS PANEL  v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11219648/OH4A6A1-1.-.ASSY.FCS.PANEL.v4.pdf)

Couldn't get trace lines on the motor driver stack.

jrsteensen, 2023-04-13:
IN BACKCHECK

[OH4A6A1-1 - ASSY FCS PANEL_-_v4-CHK1.pdf](https://github.com/jrsteensen/OpenHornet/files/11223671/OH4A6A1-1.-.ASSY.FCS.PANEL_-_v4-CHK1.pdf)

Lazarus94, 2023-04-15:
IN BACKCHECK

[OH4A6A1-1 - ASSY, FCS PANEL v13.pdf](https://github.com/jrsteensen/OpenHornet/files/11239498/OH4A6A1-1.-.ASSY.FCS.PANEL.v13.pdf)

Lazarus94, 2023-04-15:
IN BACKCHECK

Slightly updated

[OH4A6A1-1 - ASSY, FCS PANEL v15.pdf](https://github.com/jrsteensen/OpenHornet/files/11239516/OH4A6A1-1.-.ASSY.FCS.PANEL.v15.pdf)

jrsteensen, 2023-04-16:
- [ ] Delete duplicate parts list on Sheet 2

[image]

Lazarus94, 2023-04-16:
IN BACKCHECK

[OH4A6A1-1 - ASSY, FCS PANEL v15.pdf](https://github.com/jrsteensen/OpenHornet/files/11243029/OH4A6A1-1.-.ASSY.FCS.PANEL.v15.pdf)

jrsteensen, 2023-04-16:
Sorry, missed this

[image]

Lazarus94, 2023-04-18:
IN BACKCHECK

[OH4A6A1-1 - ASSY, FCS PANEL Drawing v18.pdf](https://github.com/jrsteensen/OpenHornet/files/11264218/OH4A6A1-1.-.ASSY.FCS.PANEL.Drawing.v18.pdf)

jrsteensen, 2023-04-21:
Add Balloons 

[image]
[image]
[image]

Lazarus94, 2023-04-22:
IN BACKCHECK

[OH4A6A1-1 - ASSY, FCS PANEL v18.pdf](https://github.com/jrsteensen/OpenHornet/files/11300456/OH4A6A1-1.-.ASSY.FCS.PANEL.v18.pdf)

jrsteensen, 2023-04-27:
Please incorporate and close #707 and update drawing. Backcheck looked good, but want to look at final drawing after update before signoff.

Lazarus94, 2023-05-07:
IN BACKCHECK

[OH4A6A1-1 - ASSY, FCS PANEL v22.pdf](https://github.com/jrsteensen/OpenHornet/files/11414098/OH4A6A1-1.-.ASSY.FCS.PANEL.v22.pdf)

jrsteensen, 2023-05-07:
Added to RC Release Package and closed.

closed 2023-05-07 by jrsteensen