jrsteensen/OpenHornet #539: Create Drawing: OH2A1A1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE

jrsteensen, 2022-09-14:
Create drawing OH2A1A1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE.

Ben-F111, 2022-10-09:
Build Note, Zero Sensor 

Add to the drawing
###Zero Sensor Tip Must be painted as per F360 - with at least 3 coats of Spray-paint. ###

the Photo Sensor we are finding is penatraiting Resin Prints, heat shrink

Painting the tip will stop the beam

jrsteensen, 2023-04-01:
IN CHECK

[OH2A1A1-1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE_-_v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11128238/OH2A1A1-1.-.ASSY.LOWER.INSTRUMENT.PANEL.STRUCTURE_-_v2.pdf)

jrsteensen, 2023-04-16:
Updated to latest structure with mods to facilitate future mechanical SARI and new discrete screen based solutions for RWR and SARI.

[OH2A1A1-1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE_-_v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11242935/OH2A1A1-1.-.ASSY.LOWER.INSTRUMENT.PANEL.STRUCTURE_-_v4.pdf)

jrsteensen, 2023-04-22:
Self-Checked and added to RC Release Package

[OH2A1A1-1 - ASSY, LOWER INSTRUMENT PANEL STRUCTURE_-_v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11301670/OH2A1A1-1.-.ASSY.LOWER.INSTRUMENT.PANEL.STRUCTURE_-_v6.pdf)

closed 2023-04-22 by jrsteensen