jrsteensen/OpenHornet #544: Create Drawing: OH5A2A3 - ASSY, AV COOL SWITCH

jrsteensen, 2022-09-14:
Create drawing OH5A2A3 - ASSY, AV COOL SWITCH.

Lazarus94, 2023-04-13:
IN PROGRESS

Lazarus94, 2023-04-13:
IN CHECK 

[OH5A2A3-1 - ASSY, AV COOL SWITCH v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11220557/OH5A2A3-1.-.ASSY.AV.COOL.SWITCH.v2.pdf)

jrsteensen, 2023-04-13:
Added to RC Release Package and closed.

Exprezz0, 2023-04-22:
Will need to be rechecked after I update PCB based on #625 and #594.  I'll update this once I finished fixing the PCB and F360 panels

jrsteensen, 2023-04-22:
Updated:

1. Latest PCB
2. Changed SxAL switch PN to correct OHC PN
3. Fixed title block error.

[OH5A2A3-1 - ASSY, AV COOL SWITCH_-_v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11301621/OH5A2A3-1.-.ASSY.AV.COOL.SWITCH_-_v5.pdf)

Added to RC Release Package.

closed 2023-04-22 by jrsteensen