jrsteensen/OpenHornet #545: Create Drawing: OH5A2A4 - ASSY, RADAR ALTIMETER GAUGE

jrsteensen, 2022-09-14:
Create drawing OH5A2A4 - ASSY, RADAR ALTIMETER GAUGE.

Lazarus94, 2023-05-14:
IN PROGRESS

Lazarus94, 2023-05-19:
IN CHECK

I will fix the trail lines on Sheet 1 just want to make sure everything else is fine. May also need to add an additional detail view of the assembly

[OH5A2A4-1 - ASSY, RADAR ALTIMETER GAUGE v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11517478/OH5A2A4-1.-.ASSY.RADAR.ALTIMETER.GAUGE.v3.pdf)

jrsteensen, 2023-05-19:
IN BACKCHECK

- [ ] Break title between TITLE_1 and TITLE_2.
  [image]
- [ ] What does Indicator Type A note apply to?
- [ ] Fix trail lines throughout (just noting so it isn't forgotten.)
- [ ] Move `TRIM LEADS OF NEOPIXELS...` note to local note next to balloon of FN 32 and delete sheet 3. (Images are not necessary.)
- [ ] Add to finish note for knob text to be insignia white.
- [ ] Add missing balloons for FN 20, 23, 31.

Lazarus94, 2023-05-19:
IN BACKCHECK

Do you think additional views are required or fine as is?

[OH5A2A4-1 - ASSY, RADAR ALTIMETER GAUGE v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11518993/OH5A2A4-1.-.ASSY.RADAR.ALTIMETER.GAUGE.v4.pdf)

jrsteensen, 2023-05-20:
Extra view can't hurt.

- [ ] Update to show the -24 printed gears and the 8-32 set screws.

Lazarus94, 2023-05-20:
IN BACKCHECK

[OH5A2A4-1 - ASSY, RADAR ALTIMETER GAUGE v8.pdf](https://github.com/jrsteensen/OpenHornet/files/11522711/OH5A2A4-1.-.ASSY.RADAR.ALTIMETER.GAUGE.v8.pdf)

jrsteensen, 2023-05-20:
One last super nitpicky comment:

[image]

Otherwise, perfect.

Lazarus94, 2023-05-20:
Ah whoops will fix

Lazarus94, 2023-05-20:
IN BACKCHECK

[OH5A2A4-1 - ASSY, RADAR ALTIMETER GAUGE v9.pdf](https://github.com/jrsteensen/OpenHornet/files/11522728/OH5A2A4-1.-.ASSY.RADAR.ALTIMETER.GAUGE.v9.pdf)

jrsteensen, 2023-05-20:
Added to RC Release package and closed.

closed 2023-05-20 by jrsteensen