jrsteensen/OpenHornet #548: Create Drawing: OH5A4A1 - ASSY, ELEC PANEL

jrsteensen, 2022-09-14:
Create drawing OH5A4A1 - ASSY, ELEC PANEL.

Lazarus94, 2023-04-13:
IN PROGRESS

Lazarus94, 2023-04-13:
IN CHECK

Previously drawn by Erik. Updating to current standard.

Lazarus94, 2023-04-13:
IN CHECK

[OH5A4A1-1 - ASSY, ELEC PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11220364/OH5A4A1-1.-.ASSY.ELEC.PANEL.v3.pdf)

jrsteensen, 2023-04-13:
IN BACKCHECK

[image]
[image]

Lazarus94, 2023-04-13:
IN BACKCHECK

[OH5A4A1-1 - ASSY, ELEC PANEL v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11223949/OH5A4A1-1.-.ASSY.ELEC.PANEL.v6.pdf)

closed 2023-04-13 by jrsteensen