jrsteensen/OpenHornet #549: Create Drawing: OH5A5A1 - ASSY, ECS PANEL

jrsteensen, 2022-09-14:
Create drawing OH5A5A1 - ASSY, ECS PANEL.

Lazarus94, 2023-04-16:
IN PROGRESS

Lazarus94, 2023-04-22:
IN CHECK 

[OH5A5A1-1 - ASSY, ECS PANEL v2.pdf](https://github.com/jrsteensen/OpenHornet/files/11302016/OH5A5A1-1.-.ASSY.ECS.PANEL.v2.pdf)

Exprezz0, 2023-04-25:
Saved ASSY, ECS PANEL as version 34 after updating child component dependencies for the -100 and -200 subassemblies.

Lazarus94, 2023-04-27:
IN CHECK

Exprezz0, 2023-04-30:
Needs to be updated based on issue #712

jrsteensen, 2023-05-03:
Needs updates based on #712 and have information listed in original post added to drawing covering the switch modification.

Lazarus94, 2023-05-05:
IN CHECK

[OH5A5A1-1 - ASSY, ECS PANEL v5.pdf](https://github.com/jrsteensen/OpenHornet/files/11408157/OH5A5A1-1.-.ASSY.ECS.PANEL.v5.pdf)

Lazarus94, 2023-05-07:
IN BACKCHECK

[OH5A5A1-1 - ASSY, ECS PANEL v6.pdf](https://github.com/jrsteensen/OpenHornet/files/11414087/OH5A5A1-1.-.ASSY.ECS.PANEL.v6.pdf)

jrsteensen, 2023-05-07:
Added to RC Release Package and closed.

Lazarus94, 2023-05-20:
IN BACKCHECK

[OH5A5A1-1 - ASSY, ECS PANEL v7.pdf](https://github.com/jrsteensen/OpenHornet/files/11522778/OH5A5A1-1.-.ASSY.ECS.PANEL.v7.pdf)

jrsteensen, 2023-05-20:
[image]
[image]

Note 3 should read: `FN 28, 30 THRU 33 OF -1 ASSY & FN 3 OF -300 ASSY ARE INCLUDED WITH THEIR RESPECTIVE SWITCHES, AND SHOULD NOT BE ORDERED SEPARATELY.`

Lazarus94, 2023-05-20:
IN BACKCHECK

[OH5A5A1-1 - ASSY, ECS PANEL v8.pdf](https://github.com/jrsteensen/OpenHornet/files/11522977/OH5A5A1-1.-.ASSY.ECS.PANEL.v8.pdf)

closed 2023-05-20 by jrsteensen