jrsteensen/OpenHornet #550: Create Drawing: OH5A5A2 - ASSY, DEFOG/WINDSHIELD PANEL

jrsteensen, 2022-09-14:
Create drawing OH5A5A2 - ASSY, DEFOG/WINDSHIELD PANEL.

Lazarus94, 2023-04-13:
IN CHECK 

[OH5A5A2-1 - ASSY, DEFOG PANEL v3.pdf](https://github.com/jrsteensen/OpenHornet/files/11220119/OH5A5A2-1.-.ASSY.DEFOG.PANEL.v3.pdf)

jrsteensen, 2023-04-16:
IN BACKCHECK

[image]
[image]

Lazarus94, 2023-04-16:
IN BACKCHECK

[OH5A5A2-1 - ASSY, DEFOG PANEL v4.pdf](https://github.com/jrsteensen/OpenHornet/files/11242993/OH5A5A2-1.-.ASSY.DEFOG.PANEL.v4.pdf)

jrsteensen, 2023-04-16:
Added to RC Release package and closed.

closed 2023-04-16 by jrsteensen